GridSpace/grid-apps #263: CNC "rough" operation does not retract the tool

TAR-ALEX, 2023-11-24:
Description:
CNC "rough" operation does not retract the tool and goes straight through the material in some cases. In my particular case, it actually damages the model it is cutting out.

How to replicate this:
I used version 3.9 at this link: https://grid.space/kiri/
I loaded the following model and switched to the cnc mode:  [magic.stl.zip](https://github.com/GridSpace/grid-apps/files/13455769/magic.stl.zip)
I used the default CNC machine. 
I applied the following rough operation: 
[image]
After animating the model I get the following results (damaged areas circled in red):
[image]
The movement paths confirm this:
[image]

Oddly enough, it works perfectly with a smaller bit and does all of the retractions properly.

TAR-ALEX, 2023-11-24:
After some more testing, it appears that it works as intended in version 3.8 for the same exact input.

stewartoallen, 2023-11-27:
Thanks for digging into this and providing an example. I will take a closer look when I return in mid December.

stewartoallen, 2024-06-07:
this appears to have been fixed by other recent path fixes. please confirm and re-open if you can still duplicate in 3.9.6 or 4.0

closed 2024-06-07 by stewartoallen

TAR-ALEX, 2024-07-08:
Thank you for the fix! I confirm that the problem has been solved and can no longer replicate the issue.