GridSpace/grid-apps #264: Problems slicing in FDM-mode

korpx, 2024-01-15:
Strange results when slicing in FDM-mode - Kiro:Moto 3.9

Is it a bug or a problem with my settings?

I have a simple test object with three different heights in the same model - 0.2, 0.4 and 0.6mm.

[image]

When slicing it the 0.2 and 0.4mm tall parts end up in the result, not the top layer of 0.6mm.

[image]

[slicingissue.zip](https://github.com/GridSpace/grid-apps/files/13940423/slicingissue.zip)

Zip containing;

- my_creality_ender_3.km - exported machine profile
- Ender 3 PLA.km - exported settings
- ExtrusionMultiTest1-100.stl - test object

stewartoallen, 2024-01-16:
yes, this is a bug. thanks for the report. should be easy to fix

stewartoallen, 2024-01-16:
fixed with commit 733d5b275f875fd0cf9bb79ce9889662f0fbf64a
will push to production later today

closed 2024-01-16 by stewartoallen