GridSpace/grid-apps #269: Angle of Slope Slice

Y13Lab, 2024-04-18:
In the slicer CrealityPrint, in the experimental section
there is a setting Angle of Slope Slice.
When this setting is enabled, the slicer prints at the selected angle.
In this mode, movement occurs in three coordinates x,y,z.

[image]

But there are some flaws in it that prevent it from being used.

I would like to have this functionality in slicer Kiri:Moto.
Please add this functionality to Kiri:Moto slicer.

stewartoallen, 2024-04-18:
The "belt" mode in KM (developed for the CR-30) is hard coded to 45 degrees. Do you need an angle other than that?

Y13Lab, 2024-04-18:
Belt mode uses a tilted portal.
I want to print on a regular printer.
https://www.youtube.com/shorts/tRad3QEa__U

stewartoallen, 2024-04-18:
I see. It would be far easier to write a simple script to translate the gcode output for CR-30 into traditional X,Y,Z coordinates than to modify a slicer to do this.

Y13Lab, 2024-06-09:
I don't know how to modify gcode CR-30, I do it like this.

First, the stl model is rotated to the required angle.
The rotated model is then sliced.
And apply reverse gcode rotation using script.
https://github.com/Y13Lab/Angled_Slicing

I would like to have such functionality in the slicer, so I appeal to everyone, but so far no one wants to do this

closed 2024-06-11 by Y13Lab