GridSpace/grid-apps #278: Part anchor appears under part [FDM]

Adqam64, 2024-06-08:
When slicing the STLs for a CR-30 belt printer for files from [this Thingiverse link](https://www.thingiverse.com/thing:3095799/files), the anchor appears under the part for some reason.  Verified with at least the 8x8 and smooth files.
[image]
[image]
[image]

stewartoallen, 2024-06-08:
organic parts rarely have real flat bottoms. to compensate, set the `flatten` value under `base` to something like `0.5` or `0.75` and report back :) pair that with the `lay flat` option to select a face to place on the bed. but flatten is the more important option on the belt.

Adqam64, 2024-06-09:
Ah, flatten fixed it.  Thanks!

closed 2024-06-09 by Adqam64