GridSpace/grid-apps #279: Shell width / Nozzle size >= 1.0 will generate extra moves.

Kenzu, 2024-06-22:
Hi

Pictures tell it all.

[image]
[image]

Thanks

Jesper

stewartoallen, 2024-06-23:
if you don't mind, please email the workspace (.kmz) to [ sa at grid dot space ] - will save me time synthesizing a test case. thanks

Kenzu, 2024-06-24:
I

I had to put it in a zip file :-)

[workspace.zip](https://github.com/user-attachments/files/15950496/workspace.zip)

closed 2024-06-24 by stewartoallen

Kenzu, 2024-06-25:
> completed via b8f938b

Perfect, will do some testing today.

Thanks a lot

Jesper Lindeberg

Kenzu, 2024-06-26:
Hi

I did some tests and everything seems to work :-)
I also found some good settings for my channel letters.

So thanks a lot

[image]
[image]

commit b8f938b1 (2024-06-24): fixes #279 by adapting retract distance to device nozzle diameter