GridSpace/grid-apps #281: Slow Printing Speed

goldenninja13, 2024-07-11:
For some reason after this recent update that changed the appearance of the slicer I am getting really slow printer speed. I have tried changing my minimum movement speed but it seems like it is still overriding that value. If I had to guess it prints at about 1/10 of a mm/s. I'm using an ender 3 v2. It wasn't doing this until this update.

stewartoallen, 2024-07-11:
can you attach a workspace (.kmz) that I can use to replicate? keeping the same workspace and switching between 4.0 and 3.9 produces very different gcode?

goldenninja13, 2024-07-12:
It's not allowing me to attach a (.kmz) file type in github

stewartoallen, 2024-07-12:
please email to [ sa at grid dot space ] or DM me on Discord. thanks

stewartoallen, 2024-07-12:
this looks like a rendering bug in Preview mode. in the example you provided, the rendered paths are showing up as *very* slow, while the actual gcode does not match. when I re-import the generated GCode, it's a totally different picture showing more normal extrusion rates. this particular print shows a 4.5 hour print time.

[image]

[image]

stewartoallen, 2024-07-12:
this is a good torture test. I'll use it to figure out what's going on and report back here.

goldenninja13, 2024-07-12:
Thank you. The thing I don't understand with it is it will run like normal
for a little bit but then it will change and move so slow it doesn't even
look like it's moving. So I would have to think that somewhere when
outputting the gcodes it is not sending the right values. The print I sent
you was printing like normal for the first 30 minutes and it did seem like
it was only going to take 4-5 hours but then it slowed down and has now
been printing for 80h and is only 55% complete.

stewartoallen, 2024-07-12:
aha. your output speeds are all 0, which leads to undefined behavior.

[image]

when I put in more sensible settings, things turn out better. I would also recommend using `linear`, `triangle`, or `grid` infill. Gyroid isn't helping here and takes tons more memory.

[image]

stewartoallen, 2024-07-12:
there is one remaining issue which is the model reports as 'non-manifold' which leads to a solid layer in the middle of the otherwise hollow sphere (it's in the trench area). i'll see if there is anything I can do to work around that, but fixing the model would be helpful.

goldenninja13, 2024-07-13:
I figured that was probably an issue but I did print this model before the
update and it printed completely hollow.

stewartoallen, 2024-07-13:
I've added a fix for the non-manifold slicing that will be in 4.0.6

closed 2024-07-13 by stewartoallen

goldenninja13, 2024-07-13:
Thanks for the help