GridSpace/grid-apps #284: Will not slice CNC for ShapeOKO XXL

Neanderthal50, 2024-08-02:
ver 4.0.16 worked well, 4.0.17 hangs at halfway progress.  Tried older Onshape files that worked before, same hang issue.  Cleared cache, no change.

stewartoallen, 2024-08-02:
Is this happening in Onshape or on Grid.Space? Is there a workspace you can share with me? I'm testing KM in Onshape now with 4.0.17 and can't reproduce. Thanks

Neanderthal50, 2024-08-02:
https://cad.onshape.com/documents/440b88aa532adc89737b5c21/w/ace64bd26ae4d217627b8992/e/b6e6a0cbaf312a653fa45721?renderMode=0&uiState=66ad5c43b58fb301b8545b8b

Neanderthal50, 2024-08-02:
Appears to be the drill process.

stewartoallen, 2024-08-02:
Thanks. I'm able to slice these parts in FDM and CNC modes. Can you look on the Javascript console for errors produce during slicing? Or use KM's Files -> Export menu to produce a `.kmz` file that I can use to replicate your exact setup.

[image]

[image]

stewartoallen, 2024-08-02:
found it. thanks. will push a fix to production in a couple of hours.

Neanderthal50, 2024-08-02:
[workspace_V1.zip](https://github.com/user-attachments/files/16475882/workspace_V1.zip)

In the source drawing we set the holes to 7/32 in.  I have a tool set to 0.21875 in KiriMoto.  Is there another method to set auto hole sizing/drilling/match?

Neanderthal50, 2024-08-02:
Awesome!  Thank you.  We'll continue building FRC robots at that time.  LOL

stewartoallen, 2024-08-02:
there is a "marking" option that will mark all circular holes. but I have a TODO item to allow manual selection of holes similar to the way pocket and trace work.

stewartoallen, 2024-08-02:
fix pushed. shift + reload and test. if in onshape, caching of iframes is trickier and may take some time to propagate.

commit 5d55cae9 (2024-08-02): replace center average with geometric center. fixed #284
closed 2024-08-02 by stewartoallen