GridSpace/grid-apps #285: [improvement] animation step back

tcurdt, 2024-08-10:
The current buttons during the animation look like this:

[image]

Unfortunately there is no step back.

Would that be possible to add?

I personally would change the current

```
|<- restart
||  pause
>|  step
>>  faster
```
to

```
|<- restart
|<  step back
||  pause
>|  step forward
>>  faster
```

or even

```
|<- restart
|<  step back
||  pause
>|  step forward

<<  slower
8x
>>  faster
```

stewartoallen, 2024-08-10:
`step back` would be extremely memory intensive which is why it's not currently implemented. I do like the last proposed layout for speed changes.

tcurdt, 2024-08-10:
Ah, bummer.