GridSpace/grid-apps #299: add SVG import dialog

stewartoallen, 2024-08-30:
that offers / persists options for scale, boolean interpretation, and extrusion values

tcurdt, 2024-08-30:
Import from inkscape seems to work as expected as it exports the width and height in mm

```
<svg
   width="100mm"
   height="100mm"
   viewBox="0 0 100 100"
```

not every svg has this though. Affinity designer asks for a dpi (default is 300) and then exports

```
<svg width="100%" height="100%" viewBox="0 0 3508 4961"
```

Currently the only workaround is to export as 25.4 dpi

```
<svg width="100%" height="100%" viewBox="0 0 297 420"
```

So I guess it's important to ask for the dpi if no units are provided inside the document.

stewartoallen, 2024-09-10:
import options coming today

[image]

closed 2024-09-10 by stewartoallen

tcurdt, 2024-09-10:
But there is no dpi. Wouldn't that still be a problem with:

```
<svg width="100%" height="100%" viewBox="0 0 3508 4961"
```

stewartoallen, 2024-09-10:
that svg provides no units/scale reference so units are assumed to be mm

tcurdt, 2024-09-10:
But that was exactly the problem I was having.

The export dpi affects the viewBox and by that the assumed mm.
I guess one could argue this the problem of the exporter. Not sure.

But it feels like having an option for some scaling on import can't hurt?

stewartoallen, 2024-09-10:
I can add a n`override units` pull-down dialog on import that has options like: `mm`, `inch` that allows for interpretation of coordinates in a specific unit space when not provided. but that doesn't help "pixels" which has no meaningful interpretation. in which case a `scale` dialog option is possible. but that just means you're experimenting but importing, seeing how far off it is and repeating the import process with different #s. to be really accurate for parts, your exporter either needs to provide units or you need to scale assuming mm

tcurdt, 2024-09-10:
Let me try again...

The default export with Affinity is 300 dpi. That exports:

```
<svg width="100%" height="100%" viewBox="0 0 3508 4961"
```

That's pixels without explicit units.
Overriding the units is not enough.
The desired size is not 3508mm x 4961mm.

If I set the dpi for the export to 25.4 the viewBox will be in mm.

```
<svg width="100%" height="100%" viewBox="0 0 297 420"
```

As the relationship between mm and pixel is defined by the dpi there should be no experimenting.

3508 pixels at 300 dpi means 11.69333333 inch. That's 297mm.
The factor of 25.4 is based on the metics vs imperial conversion and fix.

As long as my SVG export is always 300 dpi it should be a fix factor.
IMO that is a bit of hack though.

I personally find the notion of dpi a little flawed for vector graphics - but it is what it is.
I can see how a scale factor of 0.03937008 would be a little weird.

Specifying the dpi for files without units feels more in line what Adobe and Affinity have in mind there.
It maps the viewBox pixels to an real size - unless the units are given of course.

Ideally the import dialog would allow to specify the "dpi for SVGs without units".
So it becomes:

```
width = hasUnits(svg.width) ? svg.width : svg.viewBox.width/dpi*25.4
```

Does that make more sense now?

stewartoallen, 2024-09-10:
yes. i'll add a checkbox for `use dpi` that allows for this behavior when explicit units are missing. will push that update in a couple of hours.

commit 2239af15 (2024-09-10): add svg import dialog with options fixes #299