GridSpace/grid-apps #302: unable to select tool on small monitors

daduke, 2024-09-16:
Hi,

just started learning kiri:moto and so far I really like it. Thanks for a great tool!

I ran into a small UI issue when using kiri:moto on my 13" laptop: I have a rather long list of tools in CNC mode, and when this list reaches the bottom of the screen, it's not possible to select a different tool since the tool drop down just disappears when moving the mouse. I have to CTRL+- the browser viewport until the dropdown is small enough to fit on the screen.
Hopefully this screenshot can give the idea:
[image]
I guess adding a scroll bar to the tool list would solve the problem.

thanks,
-Christian

stewartoallen, 2024-09-16:
which browser and OS is this? I just tested in Chrome and Firefox and the tool drop down is scrollable in both of those.

daduke, 2024-09-16:
this is FF 130 on Fedora 40. You are correct in that on Chromium I get a scrollable tool list. On closer inspection FF also shows a very faint scrollbar, but it's not working correctly. So it might be a browser bug instead? thanks!

closed 2024-09-16 by stewartoallen