GridSpace/grid-apps #303: Pocket away all stock from contour job?

paukstelis, 2024-09-22:
I'm looking for a way to remove all the remaining stock following a contour operation (or perhaps before). The idea being to leave the relief of the contour job on its own. I'm not sure if there is currently a way to do this, but it would seem making the stock faces/edges selectable for pocketing might do it? I suppose that would also require knowing the intersection of the object and that stock face. I can achieve it by modifying the STL to have a thin plane the same dimensions as the stock, but this requires an additional step that would be simpler by adjusting the stock size.
[workspace_Dekor_02_09.kmz.zip](https://github.com/user-attachments/files/17090478/workspace_Dekor_02_09.kmz.zip)

stewartoallen, 2024-09-23:
I'll think about other ways to accomplish this as I've just started the 4.1 dev branch.

paukstelis, 2025-11-28:
My work around has been to make an SVG that is the size of the stock and import it with a very small Z-height. Then merge the meshes.