GridSpace/grid-apps #305: 1/8" ball end cutter looks to large

dshier65, 2024-09-27:
on CNC mode the cutter appears too large.   
[image]

stewartoallen, 2024-09-27:
can you share a screenshot of your tool definition?

dshier65, 2024-09-27:
I did add a screenshot.  Is it not showing up?  I will do another one when
I get back to the computer
Dan

dshier65, 2024-09-27:
Yes tool definition I will add that

dshier65, 2024-09-27:
is this what you are looking for?
[image: image.png]

dshier65, 2024-09-30:
is this what you are looking for>
[image]

stewartoallen, 2024-09-30:
yes. you need to either check the `metric` box so `3.175` is interpreted as `mm` *or* change `3.175` to `0.125` which is the size in inches

dshier65, 2024-09-30:
Ok, thanks for the clarification.

closed 2024-09-30 by stewartoallen