GridSpace/grid-apps #309: [FR] Option to random placement of print

Kenzu, 2024-10-03:
Hi

Would be cool to have a option to place your print random on the buildplate, to "even" wear of the buildplate.

Thanks

stewartoallen, 2024-10-03:
ensure `free layout` is enabled and optionally uncheck `auto layout` if you don't want parts to re-arrange on import. you should be able to place parts anywhere on the build plate.

[image]

Kenzu, 2024-10-03:
Hi

It’s not because I want free placement, but I want the slicer do random placement for every new print/slice.

stewartoallen, 2024-10-03:
I see. This is an option in belt printing mode, but I can see where it's useful for standard printing as well. Would it suffice to add a button to "randomize arrangement" on the build surface?

Kenzu, 2024-10-03:
That would be perfect. I see no need for more options. :-)