GridSpace/grid-apps #312: Outline seems to ignore some geometry...

Harvie, 2024-10-08:
[chaos-cam.stl.zip](https://github.com/user-attachments/files/17299464/chaos-cam.stl.zip)

when i do roughing and outline (X axis) on provided model, resulting g-code seems to leave unfinished geometry with just roughed finish:

[image]

Harvie, 2024-10-08:
Just realized it can be mitigated by lowering/changing the stepover... It's probably not even a bug.
Weird thing is that roughing has same stepower, but manages to get there...

stewartoallen, 2024-10-09:
can you provide a `.kmz` workspace usings files --> export? contouring slices the part in a completely different way than roughing.

Harvie, 2024-10-09:
[workspace_chaos-cam.kmz.zip](https://github.com/user-attachments/files/17311329/workspace_chaos-cam.kmz.zip)

yeah. i think it's more like feature than bug... contour just moves in pre-defined lines given by the stepover, while roughing works with offsets of the mesh geometry, that's where the difference comes from...

closed 2025-02-07 by Harvie