GridSpace/grid-apps #321: invalid exclude of pocket

tcurdt, 2024-12-15:
If a tool is too big for a pocket operation, the operation is skipped.

But here the tool is 3.175mm and the hole to pocket is 3.3mm.
[image]
[image]

Yet, the operation is excluded. IMO that's a bug.

[bug.kmz.zip](https://github.com/user-attachments/files/18141813/bug.kmz.zip)

stewartoallen, 2024-12-15:
this hole is highly faceted. the poly offsetting library concludes that it's zero in size when offset. reducing the tool diameter to 3.165 from 3.175 succeeds. I am making changes to allow offset polys close to 0 to work.

[image]

[image]

tcurdt, 2024-12-15:
While any improvement is of course great, I think just [reporting on the problem](https://github.com/GridSpace/grid-apps/issues/320) would go a long way.

As discussed on the chat, I can confirm a higher resolution includes the hole in the op, but it seems the offset is too small to spiral into.

[image]

tcurdt, 2024-12-15:
I tried a bit more.

Even with much higher resolution the current dev version still fails to pocket a 3.2mm pocket with a 3.175mm endmill.

[image]

stewartoallen, 2024-12-15:
in the clipper library, when the poly is offset inward, close points are collapsed. this may be tunable to some degree. but this outcome is not entirely surprising.

tcurdt, 2024-12-15:
So IIUC it's the algo from clipper that optimizes away some points to calculate the area.
And the optimization leads to a simplification that leads to a smaller area and hence gets excluded.
Did I understand that correctly?

Any other strategy that could work?

stewartoallen, 2024-12-16:
essentially. the right fix is updating the drill op. trace will have the same problems.

stewartoallen, 2025-07-21:
the new drill op in 4.2 and 4.3 handles this when it's less faceted

closed 2025-07-21 by stewartoallen