GridSpace/grid-apps #322: [FR] restrict operations to a max depth

tcurdt, 2024-12-16:
Use case:
I want to cut the first x mm with a downcut bit and then switch to an upcut bit.

It seems easiest to handle this in two separate operations.
But I would somehow need to restrict the max depth of the first operation.
The second operation could then do the full depth.

(And ideally, with rest machining skip the already cut parts)

stewartoallen, 2024-12-16:
you can limit the cut depth with "Z Bottom" ... but there is not "Z Top" equivalent. And it's global, not per operation

tcurdt, 2024-12-17:
interesting ... could be a workaround for now ... but still would be nice to have it a little easier in the future

stewartoallen, 2025-02-08:
I've added a "Z Top" limit for release tomorrow in the 4.1 branch

tcurdt, 2025-02-08:
...also just global, right?

stewartoallen, 2025-02-08:
until I can figure out two things:

1) how to allow local overrides of global settings in each operation without massively bloating or changing the UI
2) how to visualize local overrides in each op. consider that z top and bottom show planes when set. how does that visualization take into account op local overrides

tcurdt, 2025-02-08:
If you want, I am happy yo help with making at least some suggestions for 1).

As for 2) I think it should not show a full plane (by default) but only the areas affected.
Icing on the cake: Also have a global setting that "show z top/bottom as planes" which would show all of them as planes. Just make it easier to check them.

tcurdt, 2025-02-08:
A quick sketch

[image]

The operation config has two sections. Overridable and non non-overridables. Ideally all would be overridable, but it may or may not explode the default configs on the left. Might be overkill for some. But things I would love to see overridable in every operation:

- conventional
- depth first
- z top
- z bottom
- cut through

The "select" is the current "+". And I never understood the why the "✓" button is needed.
Speaking of which: I think all operations should (optionally) allow for selecting. Be it roughing, tracing or outline.

stewartoallen, 2025-02-08:
the goal is to have selectable regions on more operations starting with drilling. pocket and trace are really two different types of selection: area vs boundary. where boundaries are not required to be closed lines. drill selection will again be different because it pre-identifies holes and shows them to you. holes could be selected as boundaries, but not easily when the hole goes through a curved or angled surface. the check mark is the "done selecting" button which you can also use the `esc` key for. but I could just as easily replace "+" with "select" and make it a toggle action

tcurdt, 2025-02-08:
> the goal is to have selectable regions on more operations starting with drilling. pocket and trace are really two different types of selection: area vs boundary. where boundaries are not required to [...]

Thumbs up to pretty much on all of that. Just some more input...

From a UX perspective I think it doesn't matter that the selecting varies in type and might be pre-selected. Keeping the approach consistent is probably a fair approach.

Looking at the overall goal in mind I think it would be nice to have:

- "+"/"select" that switches to "done" ... to switch between the two modes
- instead of the "✓" maybe have a "auto select" button. Be it roughing or drilling or any other op - it would just select what it finds and then the user could change the selection. Be it when something is not detected and missing or if the user wants to change ... select e.g. all holes and remove some from the selection. Ideally that would also support area selection.

stewartoallen, 2025-02-08:
this is the treatment I'll start with

[image]

[image]

stewartoallen, 2025-02-09:
conventional, z top, z bottom implemented in 4.1.0-D8 and ready for you to test.

stewartoallen, 2025-02-10:
given that you can now control z bottom per operation, combined with any sufficient z thru, you don't need this per operation as well. I've extended the overrides to roughing and outline as well with the latest

stewartoallen, 2025-02-13:
these features will see continued improvements, but I will close this FR as completed

closed 2025-02-13 by stewartoallen