GridSpace/grid-apps #324: Laser only does outlines

xxxajk, 2024-12-31:
Would love to see a fill option for laser.
This is basically the same as you would do for a 3d printer, which would be engraving an SVG image.

stewartoallen, 2025-06-25:
what laser and driver software are you using?

xxxajk, 2025-06-28:
Generic laser, self built X/Y table, no Z offset (manual laser distance), with a custom Marlin G-CODE flavor.
So far I have made my own processor, using a combination of tools in FreeCAD.

First, what I do, then why. Perhaps you may gain something from it.

Basically, I take a set of PCB gerber files, and scale everything to micrometers as integers, plotting it inside of FreeCAD.
Then I track across in a beam dimensional way much like a raster and compress the on/off areas as needed using RLE to save memory.
Finally I take the RLE results and convert those to G-CODE motions.

Why:
FP is inherently buggy and has all sorts of maths problems, so instead I round to the nearest micrometer and only deal with integers. +/- a few uM isn't going to matter since the laser never steps at that resolution anyway. 
The steps/mm is as defined in Marlin. Trying to use FP doesn't make any sense, instead, just use a step count.
From my own code notes:
```
# 3:
# Nothing in the freecad path utilities does this properly, I already tried it.
# It is too convoluted. It has too much manual setup to do that is not relivent.
# Path also locked up/took too long when I tried to do outlines.
# 2D offsets lock up too.
# Both are useless for laser G-Code. Because:
#
# We already have the beam width, and the stepping resolution, and this in turn
# makes our job easier, as well as any processing on Marlin (or whatever you are using).
#
# 4:
# The imported shapes are broken into step-size chunks, 
# then rescaled into whole millimeters but are actually micrometers. 
# This is to avoid floating point errors/problems.
#
# DON'T PANIC!
#
# G-Code output does the correct thing, outputting in micrometers as fractional mm
# to the exact step size of the machine.
#
```

For the laser, I have an approximation of it's height and width (unfortunately it's a rectangular beam!).
example: for a 0.04mm beam radius with 400 steps/mm, beam radius = 16, and I have another variable that I can add to this base to compensate for the rectangular area of burn.

For the distances, I know how many steps per mm I get, so it's a simple rounding once again.

To generate the G-CODE it's a simple matter of rounding to the resolution of the machine steps, laser size in X/Y, and the PWM speed, of the laser so that I don't end up with banding effects of a fast moving laser. i.e I limit the speed of the motion in order to get a consistent paint ablation.

I have not yet released the code, which is in python, but I can post it if you would like to take a look at it.