GridSpace/grid-apps #329: 4.1.0-D6

malcinator, 2025-02-04:
Just downloaded and installed (over written) on MacOS 15.3. When I try to open the app i have this error.

[image]

The app continues to show its open in the dock but clicking its icon does nothing.

stewartoallen, 2025-02-04:
fixed by f5406a69d920eac1e02cef764a703e7ece7487e6 and building new binaries now. once you confirm, please close the ticket.

malcinator, 2025-02-04:
Still getting this error....

[image]

stewartoallen, 2025-02-04:
the binary hasn't updated yet. mac build still going. apple notary takes a long time.

[image]

malcinator, 2025-02-04:
Alright, I’ll check back later.

Thanks.

Malcolm Summerton
Tel:  <>‭+33 (0) 7 85 97 93 53‬
Mail:  ***@***.***
Telegram: t.me/malcinator

stewartoallen, 2025-02-04:
though we still may not be out of the woods. the local binary runs fine. when downloaded and tun, it's being treated differently.

stewartoallen, 2025-02-04:
the current build should be good. I'm not entirely certain what's going on. I rebuilt it several times without making any code changes and several failed while one worked. will have to dig deeper later today when I have time.

malcinator, 2025-02-04:
That's fixed it. 

Thanks a bunch.