GridSpace/grid-apps #331: build fail from source code

yananlemon, 2025-02-08:
I am new to node and grid-apps, I try to build it from source code,env is as below:
1. windows 10
2. grid-apps-4.0.33
3. node version: v12.22.12,v13.14.0,v19.9.0,v22.13.1.

But when I execute npm install, they all fail with the following error message:

`grid-apps@4.0.31 preinstall
> node bin/install-pre.js; npx webpack --config bin/webpack-three.js

node:internal/modules/cjs/loader:1093
  throw err;
  ^

Error: Cannot find module 'E:\other-code\grid-apps-4.0.33\bin\install-pre.js;'
    at Module._resolveFilename (node:internal/modules/cjs/loader:1090:15)
    at Module._load (node:internal/modules/cjs/loader:934:27)
    at Function.executeUserEntryPoint [as runMain] (node:internal/modules/run_main:83:12)
    at node:internal/main/run_main_module:23:47 {
  code: 'MODULE_NOT_FOUND',
  requireStack: []
}

Node.js v19.9.0
npm error code 1
npm error path E:\other-code\grid-apps-4.0.33
npm error command failed
npm error command C:\Windows\system32\cmd.exe /d /s /c node bin/install-pre.js; npx webpack --config bin/webpack-three.js
npm error A complete log of this run can be found in: C:\Users\Administrator\AppData\Local\npm-cache\_logs\2025-02-08T01_49_21_138Z-debug-0.log
`

Any idea? Thanks.

stewartoallen, 2025-02-08:
I'm still trying to get the build to work on windows. it's a host of problems including script interpretation and sym link support (or lack thereof). I'm currently building the .exe's using electron builder on mac. wish I had a better answer.

stewartoallen, 2025-02-08:
I was able to get KM running from source on Windows by installing WSL Ubuntu, cloning the repo there. I used nvm to install / manage node versions, but you have other options for getting node into WSL. the Electron app will not build in WSL because it requires all of the gui / x / whatever libraries, which are missing in WSL. but you can start KM in dev or prod mode with:

```
npm run setup
npm run dev --or-- npm run prod
```

the Bambu integration code only runs in `dev` mode

stewartoallen, 2025-03-02:
this is how I got builds to work under windows using PowerShell:

this command allows `npm` to run once it's installed:
```
winget install --id Git.Git -e --source winget
```

install git:
```
winget install --id Git.Git -e --source winget
```

install node / npm:
```
winget install Schniz.fnm
fnm install 22
node -v # Should print "v22.14.0".
npm -v # Should print "10.9.2".
```

clone grid-apps repo:
```
git clone https://github.com/gridspace/grid-apps
```

build KM binary
```
npm run setup
npm run build-win
```

`.exe` will be deposited in the `dist` directory

closed 2025-03-02 by stewartoallen