GridSpace/grid-apps #334: Tabs feature broken

marsiancba, 2025-03-01:
Seems there is some problem with tab rotation. When I add tabs: 

[image]

then restart the app, they lose angle and are always rotated along the X axis.

[image]

This seems to affect also slicing.

Version 4.1.1
(I have tried to find older version where this works, but the version switcher on the page does not work atm)

stewartoallen, 2025-03-01:
verified. I'll try to fix it for this weekend's release

closed 2025-03-01 by stewartoallen

marsiancba, 2025-03-01:
The restart bug is fixed, but slicing is still broken. There seems to be the same bug there - it is slicing as if all tabs are oriented along X axis:

[image]

stewartoallen, 2025-03-01:
confirmed. looking into it

stewartoallen, 2025-03-01:
fixed in production

marsiancba, 2025-03-02:
I have just found another place - Flip operation:

[image]

commit e65c6e1d (2025-03-01): fixes #334 ensure tab rotation quaternions are stored as arrays