GridSpace/grid-apps #343: CNC - Toolpath - Slice show - Selection?

Monberg75, 2025-04-07:
So, I'm wondering why I'm not able to separate the toolpaths for closer inspection in either this or in preview window?

In this way, I can't tell what is what, and how it behaves, as it is all on top of eachother.

[image]

stewartoallen, 2025-04-07:
ctrl + click (command + click on mac) on an operation will toggle it enabled / disabled making individual inspection of paths easier. you can also ctrl + shift + click to XOR toggle all operations to the opposite statea of the one clicked.

Monberg75, 2025-04-09:
Ok, that is internal knowledge. Maybe an eye icon along with the trash can
icon on the toolpath would be more user-friendly. But thank you for
telling. :)

bakedPotatoLord, 2025-07-21:
This control is viewable in the [docs](https://docs.grid.space/kiri-moto/controls#cam-controls).

fixed by [this commit](https://github.com/GridSpace/grid-apps/commit/d2ed27708406f70fb1a40c563ca649907a8f258f)