GridSpace/grid-apps #352: Large tool support / animator issue?

MROLLS1, 2025-04-25:
I have noticed that creating a common 40mm router surfacing bit (1/4 inch shank) results in some strange behaviour in KM. 

When the part origin is on the top of the stock, southwest corner for XY 0,0, I generate a toolpath to simply surface the top of the stock to remove 1mm of material across the entire space. It slices fine, preview looks good, but then when I animate it, the tool suddenly appears shifted over to something like X+15 Y+15 and it commences the operation from that point, leaving uncut stock and cutting way off over the far side of the part. 

However, if I export the gcode to my CNC and execute it, the tool starts at the origin as expected and follows the path as expected. 

Note; this issue does not affect small "common" size router bits, like a 1/4" end mill. I do not know where the breakpoint is but it seemed to have the same error at a 20mm diameter, it may be worth a sweep to see what the maximum diameter cutting bit is before it starts to break away from the origin as the tool start point? 

Attached .zip including the KMZ, as Github won't allow kmz upload. 

[KM Facing Test Large Tool.zip](https://github.com/user-attachments/files/19910561/KM.Facing.Test.Large.Tool.zip)

MROLLS1, 2025-04-26:
https://forum.grid.space/t/animation-erroneously-offset-only-above-certain-tool-diameter/2219/4

I think this is definitely related to this post as well...

bakedPotatoLord, 2025-05-05:
I tried animating this KMZ, and was able to reproduce. This artifact was created:

[image]

stewartoallen, 2025-06-24:
fixed with a recent update

commit 7b1f230d (2025-06-23): fix numerous bounds issues with indexed stock fixes #340
fix state variable updates with indexex stock
fix large tool animation fixes #352
closed 2025-06-24 by stewartoallen