GridSpace/grid-apps #354: Trace operation not generating toolpath

MROLLS1, 2025-04-26:
I've made a test block with some holes and pockets, I've attached a ZIP folder with the KMZ file.

I was attempting to use the Trace tool to trace the top rim of the pockets with a 90 degree v-bit, but when I select the edges to trace, and set all the parameters up for the trace op, it does not generate any toolpath. 

Am I doing something wrong or is this a bug?

[V-groove trace test.zip](https://github.com/user-attachments/files/19922544/V-groove.trace.test.zip)

MROLLS1, 2025-04-26:
It may be worth noting that Kiri:moto just became unresponsive and will not reload. Perhaps it wouldn't generate toolpaths because something was wrong elsewhere?

bakedPotatoLord, 2025-05-01:
Thanks for the bug report. 

It seems that this error is caused by having `step down` set, but a `cut through` parameter of 0. When `cut through` is set to 2, everything seems to behave fine. feel free to take a look at this KMZ to make sure it's what you were expecting. 
[workspace_v-groove_trace_test.zip](https://github.com/user-attachments/files/20002669/workspace_v-groove_trace_test.zip)

stewartoallen, 2025-05-01:
cut thru is a good workaround if you are able to select the desired edge. but that's difficult. the trace op as currently constructed doesn't work well with this part (it's too compute intensive and mis-identifies real edges). we will need to move to a proper edge selection technique (based on the widget trace-edges algorithm) vs the current "slice" technique to rectify this.

bakedPotatoLord, 2025-07-21:
The 4.2 trace overhaul fixes this.

closed 2025-07-21 by bakedPotatoLord