GridSpace/grid-apps #360: dragknife adding incorrect paths

stewartoallen, 2025-05-05:
near straight lines are inducing curves

[old_bay.kmz.zip](https://github.com/user-attachments/files/20031050/old_bay.kmz.zip)

stewartoallen, 2025-05-09:
simpler test case

[drag knife error.kmz.zip](https://github.com/user-attachments/files/20124725/drag.knife.error.kmz.zip)

bakedPotatoLord, 2025-05-12:
The abnormalities seem to be occurring at the vertices of the mesh.
[image]

stewartoallen, 2025-05-12:
I wonder if this is an angular wrap-around issue. like 0.1 degrees to 359.9 degrees when the actual separation is 0.2 degrees but math says 359.8 ... there is a function that fixes this somewhere in the code. but the algo is also quite simple. angular deviations over some threshold trigger the drag arc injection.

bakedPotatoLord, 2025-05-12:
Some of the slopes are being set to  `0` or `180` when they shouldn't be. when they are, the tight angle detection is triggered on both sides of that slope. My best guess is that the slope detection code has a bug somewhere. I'm not getting anywhere with this right now, so I'm going to put it on the back burner and come back to it.