GridSpace/grid-apps #388: climb / conventional checkbox sense is inverted

foxharp, 2025-06-15:
I'm running the linux appimage version 4.1.9.

I have a roughing pass, and an outline pass.

Roughing seems to do all of its steps in conventional (CCW for an interior cut) direction.

If I don't check the "Conventional" box for the outlining pass, it cuts the same way as roughing did.  If I check the Conventional box, the direction switches to CW.

Can provide export .kmz if needed.

foxharp, 2025-06-17:
Yes, your repro zip reproduces the problem.   TBH, I can't tell what direction the animation is going in Rough, because I can't make it slow enough.  Not sure why not.  Outline I can slow down.

foxharp, 2025-06-17:
If I turn on Roughing for the outside, somehow it slows everything down. (?!)  (I have feed set to '1', and animation set to '1x'.)   

In any case, with roughing turned on for both inside and outside, it's clear that climb/conventional alternates for the internal clearing levels -- probably as a side-effect of optimizing movement by going outward/inward/outward.   Without the checkbox checked (i.e., climb selected), the first pass of the inside and all of the passes for the outside, are done in climb.  Checking the box reverses the first pass of the inside, and all of the passes for the outside.

So this supports your hypothesis that it's a bug in outlining only.

stewartoallen, 2025-06-18:
@bakedPotatoLord agree this is leagacy code. can remove from `config.js` as well.

bakedPotatoLord, 2025-06-27:
Merged. deploying to prod this weekend