GridSpace/grid-apps #414: Rough inside only uses G0 for cutting moves

bakedPotatoLord, 2025-08-02:
When jumping from one poly to another, the Rough operation uses a G0 with a Feedrate.

[cribbage _board v2.txt](https://github.com/user-attachments/files/21561034/cribbage._board.v2.txt)

stewartoallen, 2025-08-03:
if the move distance is less than the tool width this should be ok. is the issue just the feed rate?

bakedPotatoLord, 2025-08-03:
That might be causing it. I was surfacing a part using a 1.5" Face mill. 

The machine I was using does not move at the F speed for G0s, and just goes as fast as possible.

I'll do some more digging.

bakedPotatoLord, 2025-08-03:
here's a KMZ for reproduction

[cutting G0.zip](https://github.com/user-attachments/files/21566269/cutting.G0.zip)

bakedPotatoLord, 2025-08-05:
I think I broke this code in the transition to arcs. iirc, the rough clearing didn't use `p2pEmit` and `polyEmit` initially, but it now does. If we want all polys to eventually be going through `polyEmit`, the function probably needs to become more robust.

stewartoallen, 2025-08-05:
I may need to wholesale refactor cnc path output sooner rather than later. as you say, the arc code may have changed things subtly in unsafe ways.